i3detroit/custom-mqtt-programs #13: fablab atmosphere sensor overflow stuff

mtfurlan, 2018-01-14:


mkfink, 2018-01-24:
Overflowing:
tele/i3/inside/fablab/sensor-cluster/bmp280 {"Temperature":24.63, "Pressure":98997.10}
tele/i3/inside/fablab/sensor-cluster/lux {"Lux":0}
83647 {"Temperature":2147483647.2147483647, "Humidity":2147483647.2147483647}

mkfink, 2018-01-29:
The office thermostat has the same overflow error.

[image]

mtfurlan, 2018-01-29:
Oh good. Does someone else want to look at my code because I'm apparently
incapable of dealing with pointers?

mkfink, 2018-02-04:
atmosphere sensors haven't had any overflows since changing to bme280 sensors, but the office thermostat is still overflowing. Example:
`2018-02-03 16:37:45.241 tele/i3/inside/office/thermostat/bme280 {"Temperature":17.94, "Pressure":83647, "Humidity":2147483647.2147483647}`

mtfurlan, 2018-03-08:
@mkfink Is the thermostat still having this issue?

abzman, 2018-03-19:
@mkfink has this issue been solved?

mkfink, 2018-03-21:
Yeah, it's been fixed since mark fixed the ftoa function we were using.

closed 2018-03-21 by mkfink