i3detroit/custom-mqtt-programs #28: Thermostat only reports rounded temperature

mkfink, 2018-01-24:
When reporting the temperature measured by the bmp280 over mqtt, the value is almost always rounded to the nearest whole number but then with two decimals of precision. For example, it only reports 25.00, 25.00, ..., 26.00, ... etc.

[image]

mtfurlan, 2018-01-24:
Oh yeah cause evan only wanted to display whole numbers on the display and
then I used that for the reporting.

closed 2018-01-26 by mtfurlan

mkfink, 2018-01-27:
It sometimes still reports out decimal fractions from halves to fifths, i.e 25.33, 24.80, etc. That suggests there might be something amiss. Will that cause any overflows?

mtfurlan, 2018-01-27:
I'm seeing stuff like .4 and .62. I'm not worrying.