i3detroit/custom-mqtt-programs #37: Light switch triggers along with ceiling fans

mkfink, 2018-02-11:
I powered the ceiling fans off via home assistant and the commons light switches east (ones by the front door) registered as row 1 having being pressed. It did it twice in this log and once afterward but it's not fully repeatable.

`2018-02-11 00:24:08.393 cmnd/i3/inside/commons/ceiling-fans/POWER OFF
2018-02-11 00:24:08.440 stat/i3/inside/commons/ceiling-fans/RESULT {"POWER":"OFF"}
2018-02-11 00:24:08.442 info/i3/inside/commons/light-switches-east/pressed eastOff
2018-02-11 00:24:08.487 cmnd/i3/inside/lights/001/POWER 0
2018-02-11 00:24:08.490 stat/i3/inside/commons/ceiling-fans/POWER OFF
2018-02-11 00:24:08.492 cmnd/i3/inside/lights/002/POWER 0
2018-02-11 00:24:08.494 cmnd/i3/inside/lights/003/POWER 0
2018-02-11 00:24:08.496 stat/i3/inside/lights/001/RESULT {"POWER":"OFF"}
2018-02-11 00:24:08.498 cmnd/i3/inside/lights/004/POWER 0
2018-02-11 00:24:08.501 stat/i3/inside/lights/004/RESULT {"POWER":"OFF"}
2018-02-11 00:24:08.549 stat/i3/inside/lights/003/RESULT {"POWER":"OFF"}
2018-02-11 00:24:08.554 stat/i3/inside/lights/001/POWER OFF
2018-02-11 00:24:08.560 stat/i3/inside/lights/002/RESULT {"POWER":"OFF"}
2018-02-11 00:24:08.565 stat/i3/inside/lights/004/POWER OFF
2018-02-11 00:24:08.592 stat/i3/inside/lights/003/POWER OFF
2018-02-11 00:24:08.599 stat/i3/inside/lights/002/POWER OFF
2018-02-11 00:24:12.203 cmnd/i3/inside/commons/ceiling-fans/POWER ON
2018-02-11 00:24:12.235 stat/i3/inside/commons/ceiling-fans/RESULT {"POWER":"ON"}
2018-02-11 00:24:12.239 stat/i3/inside/commons/ceiling-fans/POWER ON
2018-02-11 00:24:15.230 cmnd/i3/inside/commons/ceiling-fans/POWER OFF
2018-02-11 00:24:15.303 stat/i3/inside/commons/ceiling-fans/RESULT {"POWER":"OFF"}
2018-02-11 00:24:15.307 info/i3/inside/commons/light-switches-east/pressed eastOff
2018-02-11 00:24:15.316 stat/i3/inside/commons/ceiling-fans/POWER OFF
2018-02-11 00:24:15.353 cmnd/i3/inside/lights/001/POWER 0
2018-02-11 00:24:15.375 cmnd/i3/inside/lights/002/POWER 0
2018-02-11 00:24:15.378 stat/i3/inside/lights/001/RESULT {"POWER":"OFF"}
2018-02-11 00:24:15.386 cmnd/i3/inside/lights/003/POWER 0
2018-02-11 00:24:15.388 stat/i3/inside/lights/001/POWER OFF
2018-02-11 00:24:15.390 cmnd/i3/inside/lights/004/POWER 0
2018-02-11 00:24:15.393 stat/i3/inside/lights/004/RESULT {"POWER":"OFF"}
2018-02-11 00:24:15.395 stat/i3/inside/lights/003/RESULT {"POWER":"OFF"}
2018-02-11 00:24:15.398 stat/i3/inside/lights/002/RESULT {"POWER":"OFF"}
2018-02-11 00:24:15.400 stat/i3/inside/lights/004/POWER OFF
2018-02-11 00:24:15.403 stat/i3/inside/lights/003/POWER OFF
2018-02-11 00:24:15.404 stat/i3/inside/lights/002/POWER OFF`

mkfink, 2018-02-12:
> 2018-02-11 07:35:48.846 stat/i3/inside/commons/ceiling-fans/RESULT {"POWER":"OFF"}
> 2018-02-11 07:35:48.849 info/i3/inside/commons/light-switches-east/pressed eastOff
> 2018-02-11 07:35:48.868 stat/i3/inside/commons/ceiling-fans/POWER OFF
> 2018-02-11 07:35:48.882 cmnd/i3/inside/lights/001/POWER 0
> 2018-02-11 07:35:48.887 cmnd/i3/inside/lights/002/POWER 0
> 2018-02-11 07:35:48.891 cmnd/i3/inside/lights/003/POWER 0
> 2018-02-11 07:35:48.932 cmnd/i3/inside/lights/004/POWER 0

abzman, 2018-03-19:
seems to be the welder, commons switches east moved to front door controller 12v battery, see if problem persists on it

mtfurlan, 2020-01-26:
Adding caps fixed this, correct?