i3detroit/custom-mqtt-programs #52: Thermostat v1.1 changes

mtfurlan, 2018-03-22:
* All LED resistors pullup or pulldown
* Consider flipping ESP if it has wifi issues what with the radio being next to a ground plane

mtfurlan, 2018-03-23:
previous things done

* Surface mount resistors
* caps across inputs maybe?

abzman, 2018-04-03:
an extra board could be made to handle more complicated HVAC systems that use things like 2-stage cooling, 2-stage heating, heat pumps, auxiliary heating, humidifier, dehumidifier, etc...

abzman, 2018-04-23:
need 5v output to relays instead of 3.3v

mtfurlan, 2018-05-28:
Mounting holes
Extra vias next to all DIP packages for easier testing/routing modification
Logos!

mtfurlan, 2018-06-23:
V2 DONE

V3 changes
Side button and LED spacing is not consistent
Label bottom connectors
Clean up silkscreen

mtfurlan, 2018-06-23:
Right angle LEDs to close to edge

mtfurlan, 2018-06-23:
Silkscreen conventions to show pin1 for logic chips

mtfurlan, 2018-06-23:
Flip ESP

mtfurlan, 2018-06-23:
Mounting holes not square

mtfurlan, 2018-06-24:
Rename reset -> run

mtfurlan, 2018-06-24:
Led footprints
Right angle leds

mtfurlan, 2018-07-01:
Front LEDs instead?
Solder jumpers for mcp breakout address lines.
Open source hardware logo and recheck terms

mtfurlan, 2018-08-06:
Address jumpers

mkfink, 2018-09-30:
Add restart button

mtfurlan, 2019-03-24:
Silkscreen labels for connectors