i3detroit/custom-mqtt-programs #63: atmosphere board v2

mtfurlan, 2018-06-24:
Power led
Logo not under USB
Label pins silk
Copper fill under antenna
Barrel jack footprint, labeled
Check holes square
Ask Mike about size of board, look for easy enclosures

mkfink, 2018-09-09:
Swap esp to other side of PCB to test if it gets better wifi reception

mkfink, 2020-12-03:
@mtfurlan what kind of barrel jack did you want to add? I'm finding I'd like to add a 2.5mm TRRS jack to this board (for like the temperature probe they sell for the Sonoff TH line). Also an extra I2C 4-pin footprint (to fit like JST-PH) for additional modular I2C sensors. Maybe two footprints, one 2.54mm pitch and the other 2mm pitch. Could put them close together to save space and just use one or the other. I'm using an adafruit STEMMA soil sensor which natively takes JST-PH 2mm, but I have a lot more 2.54mm stuff.

mtfurlan, 2020-12-03:
What about the [sparkfun qwiic](https://www.sparkfun.com/categories/399)? Is JST SH 1mm pitch.

mkfink, 2020-12-03:
Oh those are cute, I guess maybe one of those too?

abzman, 2020-12-13:
I want to make the power headers just 0.1" spacing and have an additional option for screw terminals (possibly) Also move the sensors so they don't overlap the mounting holes

mkfink, 2021-01-21:
I'm experimenting with enclosures for the v1 board and it's giving me a few ideas.
* Move the d1 mini so that the USB port is at the edge of the board. 
* Arrange the sensor boards so their actual sensor elements aren't too close to the edge of the board so we don't need holes in the edge of the lid.
* Keeping the board single sided will make for more compact enclosures, but can manage either way.

[image]