i3detroit/custom-mqtt-programs #64: thermostat code

mtfurlan, 2018-06-24:
Redraw screen on connection
Something with fan forced sticking on

mkfink, 2018-07-03:
Fan state does not report correctly. Requested fan state appears to mirror active fan state.

mtfurlan, 2018-07-03:
Mqtt reset/run command.
Mqtt set reset interval

mtfurlan, 2018-07-05:
Delay fan turn on a few seconds.

mkfink, 2018-07-05:
publish a status (like "../thermostat/timeout true") when the inactivity timeout has been reached and thermostat is resetting to default values.

mtfurlan, 2018-07-06:
Done:
- Screen redwawing, kinda. Will need to be redone later maybe.
- Fan forced sticking on: Was a logic issue and maybe also a relay issue. Fixed now.
- Fan state reporting properly.

Todo
- Split out some functions and write some tests
- thermostat/timeout true for timeout reset
- mqtt run command
- mqtt timeout interval set; eeprom.
- Delay fan after heat/cool somehow.

mtfurlan, 2018-07-08:
Sorted todo
- Delay fan after heat/cool somehow.
- Split out some functions and write some tests
- Not fail if mqtt can't connect
- timeout interval set from mqtt
- publish stat/thermostat/run with `mqtt`/`button`/`timeout`
- mqtt run command

mtfurlan, 2018-08-04:
Delay fan by 30s done.
Kevin is now responsible for the tests, but it seems to work.

Autodetect BME vs BMP 280
Detect if temperature sensor unplugged while running

mkfink, 2018-08-20:
Changing swing changes mode to heat (observed when thermostat was set to cool and was actively cooling)

mtfurlan, 2018-08-22:
Issue with swing is known, fix is implemented.
Unresolved issue: Swing can take letters and thinks it gets 0 and this is bad.

mkfink, 2018-08-22:
Swing should also probably never be allowed to be 0 lest we end up switching HVAC 100 times a day.

mtfurlan, 2018-09-03:
Swing issue fixed;
String parsing issues fixed
Delay fan thing done
Timeout set from MQTT, but not set in eeprom

New todo
* test new reset eeprom code
* publish stat/thermostat/run with mqtt/button/timeout
* mqtt run command
* timeout not set in mqtt
* detect if bme unplugged during runtime

mtfurlan, 2018-09-03:
Something is wrong with the eeprom, not persisting properly? Maybe it's being reset every boot?

mtfurlan, 2018-09-03:
Newerest todo
* detect if bme unplugged during runtime
* test new reset eeprom code
* if you set a target or some other control to the current, it will not acknowledge this
* publish stat/thermostat/run with mqtt/button/timeout

mkfink, 2018-09-09:
3 hours after changing mode from 'AC' to 'off' the target changes from its current value to 10. This time delay seems not very intuitive. Should we even display a target when mode is 'off'?

mtfurlan, 2018-09-09:
Probably not. Maybe a countdown to time out?

mkfink, 2018-09-09:
Timeout should be irrelevant in off mode. Just checked my home thermostat and just stops displaying target when set to off. It also remembers the last set heat and cool targets when I switch back to those from off, and I think I'd want that kind of behavior on this thermostat if I were using it at home with timeout disabled.

mtfurlan, 2018-09-09:
Makes sense.

Todo
* Remember mode setpoints, will not survive reboots, just start at reset
target
* Hide Target in off mode
* Stop time out in off mode
* Time out countdown

mtfurlan, 2018-09-17:
Added query command

mtfurlan, 2018-09-30:
Set temp with remote command
Detect BME unplugged

mtfurlan, 2018-10-09:
Make it so you can't turn on heat/cool more than once every two minutes or something.

mtfurlan, 2020-01-26:
Current state of publishing topics:
```
tele/.../sensors {"temperature":19.85, "pressure":98621.7, "humidity":42.0}
tele/.../output {"state": "off", "fan": "on"}
tele/.../input {"target":19, "mode":"heat", "fan":"on", "swing":1, "timeout":0}

stat/.../mode HEAT
stat/.../target 19
stat/.../fan on
```

Command topics:
```
cmnd/.../mode
cmnd/.../target
cmnd/.../fan
cmnd/.../query
```

`tele/.../input` being tele confused me.
Is this "correct"?

Also, todo list as I currently read it:
* Set temp with remote command
* Detect BME unplug
* Save current target in eeprom

Todo items that need research into what we meant two years ago:
* Stop time out in off mode
* Time out countdown
* Make it so you can't turn on heat/cool more than once every two minutes or something.